robaki-dev/robak #18: Splitted body

buldo, 2021-11-24:
Current body is really big.
I can split it into 2 parts in 3D Builder, print and glue together.
But I think You can design smth better. Maybe shaft or bolted joints...

robaki-dev, 2021-11-24:
Surely Robak deserves a better body! I don't like current design either. I have sth in mind and I'm going to share it as soon it's ready. Thanks for the feedback!

buldo, 2021-12-10:
@robaki-dev right now my printer is broken. New part is on the way from China. I'll try as soon as possible later this month.

buldo, 2021-12-26:
@robaki-dev I tried new design. It is good and works. But there are some issues:

- this is only part that use M2 screws. This extends BOM. (And for me this is a bit harder to find M2 in my screw collection)
- this design requires ideal bed leveling( maybe it shoud be printed in some special orientation)
- it hard to assemble

[image]
[image]

buldo, 2024-01-02:
@robaki-dev after long time I tried splitted body from 2.2.0 release.  
I printed it from PETg.  
Strange, but looks like two parts holes not fits

robaki-dev, 2024-01-02:
You're right! I'll get that fixed soon. Thanks for pointing this out.

[image]

muraj, 2024-01-03:
Yeah, I had a hard time lining this up as well.  Managed to get 4/6 screws lined up, but I think that misaligned the outside.  Anyway, looking forward to new body!

cnrd, 2024-01-03:
I also had issues getting the screws in, when I tried the 2 part body.

Looking closely at the two halves, I have found a couple of other differences where the models does not align.

[image]
[image]

robaki-dev, 2024-01-07:
[image]
Fixed the alignment of the holes. Files are now available on Printables and in the release ZIP. Thanks!

closed 2024-01-07 by robaki-dev

ziguana, 2024-01-07:
i thought it was my fault that the holes didn't line up. thank you for fixing this!