robaki-dev/robak #19: Make CHub stronger

heminei, 2021-12-05:
Very often CHub brokes. I have modified the center hole: I made it smaller. Also, I increased the wall thickness where is placed the hole.

I have attached my modified version.
[Robak - CHubs - stronger.zip](https://github.com/robaki-dev/robak/files/7655871/Robak.-.CHubs.-.stronger.zip)

robaki-dev, 2021-12-08:
Hi, thanks for fixing that!
Cause I'm currently working on the v2 I don't have any compatible v1 Robak on hand to test your design.

What about publishing this as a remix on Thingiverse? If you don't have an account or just want me to do that for you let me know. I'm not going to make any further releases of v1 so that would be the easiest way to share your upgrade.

heminei, 2021-12-08:
Hi, 
I have published a remix: [](https://www.thingiverse.com/thing:5159148)

closed 2021-12-08 by heminei