robaki-dev/robak #27: Cannot Clone git repo

cereal7802, 2022-12-12:
Looks like trying to clone the repo results in an error.

Downloading DRAWINGS/HingePins.pdf (116 KB)
Error downloading object: DRAWINGS/HingePins.pdf (04f637b): Smudge error: Error downloading DRAWINGS/HingePins.pdf (04f637b836a2dc1a01f305a50a70553f9a62fe9e067daa620919f41b51f7a4f9): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

I can download the releases fine, so there is no loss of access to the files, just that it cannot be pulled using git clone without error.

cereal7802, 2022-12-12:
Hmm...well it turns out when I download the release files and extract them, I get an error loading the files too. Super Slicer says it failed to open the file. Bambu slicer has a similar error and then says the file contains not geometry.

cereal7802, 2022-12-12:
Sorry for all of the comments, I didn't spend enough time checking things before opening the issue. All of this seems to be specific to the 2.1 release. I was able to download the 2.0 release files and they work fine.

robaki-dev, 2022-12-12:
Hi, I forgot to add more GBs of bandwidth to LFS (file hosting) again. my bad. It's done now.

Please try to download v2.1.0 release once again.

Thanks! 😉

closed 2022-12-12 by robaki-dev