robaki-dev/robak #28: Motorblock & spur gear bearing missing / incompatible

apphead, 2022-12-29:
During my build I was struggling with how the spur gear axis is "connected" to the motorblock. Intuitively, I wanted to fit a second, bigger bearing into that hole but my 5x16x5mm bearings I have left from my printers just would not fit. And it is not in the design files either - on purpose?
Anyway, I am going to adjust the design and increase the inner hole by 1mm to 16. Dunno if it is a bug or not intended to contain a bearing, but I will try it anyway. I think it relieves some pressure from the small bearing next to it.

[image]

I am a huge fan of your design and will come up with some addons / improvements over the months to come. Is there a good way to collaborate with you over Fusion360 (I'm just an enthusiast with 8 years of 3d printing experience, not a pro). Github probably not an efficient path!? If you welcome my input and know a process, let me know.
Current list of todo's / issues:
- Bellcranks with nuts
- Distance servolink to topdeck too close
- Front bumper
- LED bar: magnetic mount (attachable)
- New body  with rollcage [https://github.com/robaki-dev/robak/issues/18](url)
- Spoiler
- Attachable Gearbox Cover (as posted on Discord)
- Wheelie Roll Bar (maybe)

If there is an overlap with your current efforts, please let me know

apphead, 2022-12-30:
There is not enough space to fit a 5mm bearing. But I'm not 100% happy with the current solution. Closing for now as my suggestion is invalid.

closed 2022-12-30 by apphead

robaki-dev, 2023-01-02:
yes, that hole was not intended to contain a bearing. Its size was an arbitrary choice just to make some room for the spur gear. 
sometimes the spur gear goes in like that:
[image]