robaki-dev/robak #29: Collaboration process

apphead, 2022-12-29:
I am a huge fan of your design and will come up with some addons / improvements over the months to come. Is there a good way to collaborate with you over Fusion360 (I'm just an enthusiast with 8 years of 3d printing experience, not a pro). Github probably not an efficient path!? If you welcome my input and know a process, let me know.
Current list of todo's / issues:
- Bearing motorblock (above)
- Minor improvs: stronger battery mounts (broke off after 10mins) & adjust topdeck wire holes for my Rocket 540 V3
- Bellcranks with nuts
- Distance servolink to topdeck too close
- Front bumper
- LED bar: magnetic mount (attachable)
- New body  with rollcage [https://github.com/robaki-dev/robak/issues/18](url)
- Spoiler
- Attachable Gearbox Cover (as posted on Discord)
- Wheelie Roll Bar (maybe)

If there is an overlap with your current efforts, please let me know.

maciejwitalinski, 2022-12-30:
Gearbox cover for the 2.1 version will be much appreciated.
There is one another issue - the airflow could be better to cool down the motor. With the standard cover the motor can get too hot even with 17T gear. 

Cheers,
Makavelli

robaki-dev, 2023-01-02:
Hi! I would really like to integrate your (and other people's) work into the design. I don't think Fusion360 (non-commercial) supports collaboration but we can find a way around though.

Since I don't make many modifications to the project at this point I can save it as is and handover to you. After a week or two, when you're done with modifications, you will send it back to me and your version will become the latest one. 
This method may seem archaic, but I cannot come up with a better solution right now 😉

Modifications could be published as remixes (separate parts) or kept until next Robak 2.x release.

Some parts like bumpers and wings I would prefer to keep as optional, as they may not work for everyone. We can discuss the details later.

Many thanks for the feedback and support @apphead, really appreciate it!

apphead, 2023-01-03:
> Gearbox cover for the 2.1 version will be much appreciated. There is one another issue - the airflow could be better to cool down the motor. With the standard cover the motor can get too hot even with 17T gear.

True, I noticed that the 17.5T motor gets hot with 21T Pinion on 2S. As suggested on discord, I'll add a fan. But a better airflow would certainly be a good idea too. I put it on my todo list

> Hi! I would really like to integrate your (and other people's) work into the design. I don't think Fusion360 (non-commercial) supports collaboration but we can find a way around though.
I have Educational license and I can invite somebody to my fusion project - no idea if that makes it possible to collab, but it's worth a try!? If it is fine by you, DM your (Fusion Login-) email on discord. I go by the name of Googliola or I can send you my email if you prefer - your nickname on discord is milosz, right?

robaki-dev, 2023-01-05:
on my side it looks like this 😞:
[image]

apphead, 2023-01-05:
Hmm, bummer. Maybe editing works if I invite you?

closed 2023-01-29 by apphead