robaki-dev/robak #30: I can't clone repo with LFS data

prepor, 2023-06-16:
Hello!

During a cloning, I got this message:

> Error downloading object: DRAWINGS/HingePins.pdf (04f637b): Smudge error: Error downloading DRAWINGS/HingePins.pdf (04f637b836a2dc1a01f305a50a70553f9a62fe9e067daa620919f41b51f7a4f9): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

Is this a problem on my side or github is actually limiting this repo?

prepor, 2023-06-16:
Downloading from Releases also doesn't give me real files (stl, step), only some meta data

prepor, 2023-06-16:
Ok, dowloding beta works fine [Robak_v2.2.0-beta.zip](https://github.com/robaki-dev/robak/releases/download/v2.2.0-beta/Robak_v2.2.0-beta.zip), but not source files. And it doesn't include step files

prepor, 2023-06-16:
The direct link from here https://robaki.dev/docs/download/ isn't working either

robaki-dev, 2023-06-17:
Hi @prepor,

It's on my side. Git LFS stopped working because I didn't pay for it this month 😅
Added Robak_v2.1.0.zip in releases. It includes STEP for v2.1. Just ignore `Source Code.zip` - I cannot disable it from the release,
STEP file for v2.2 will be added when it get out of beta version.

Fixed links on https://robaki.dev/docs/download/

Thanks!

closed 2023-06-17 by robaki-dev