robaki-dev/robak #34: Main body as one piece

JendaDH, 2024-11-20:
Hi, 

is it possible to have a STL model of the body parts (bodyF & bodyR) as one combined piece for larger printers?
I can eventually join myself and link with this repo. Just wondering if that makes sense.

Thanks.
J.

robaki-dev, 2024-11-20:
Hi! Here you go. There’s also a one piece low profile body I like better because it’s much lighter. The only thing is, I haven’t nailed down the best mounting system for it yet. I’m working on it, though and it should be good to go in a few days.

I’m sharing the link to the Discord post since I can’t upload the STL directly here, hope that works for you!
https://discord.com/channels/816386800820944936/892053093095727125/1160197737753624616
[image]

low profile body:
https://discord.com/channels/816386800820944936/958706795940024350/1200798712205672548
[image]

JendaDH, 2024-11-20:
Thanks 👍J.

https://discord.com/channels/816386800820944936/892053093095727125/1160197737753624616
image.png (view on web)
https://discord.com/channels/816386800820944936/958706795940024350/1200798712205672548
image.png (view on web)

closed 2024-12-18 by robaki-dev